end2endzone/AnyRtttl #3: Incompatibility with non-blocking tone routine for ATtiny85.

JeanHenriqueAlves, 2019-08-17:
I'm trying to use this library in `blocking::playProgMem` mode, but while it sounds great and works well on its own, it also stops the tone routine I'm using from working. I'm using the following routine, it can be called by using `note(n, o)`, where 'n' is the note, from 0 to 11 and 'o' is the octave, from 0 to 10:

```
const int Output = 1;                                   // Can be 1 or 4
// Cater for 16MHz, 8MHz, or 1MHz clock:
const int Clock = ((F_CPU/1000000UL) == 16) ? 4 : ((F_CPU/1000000UL) == 8) ? 3 : 0;
const uint8_t scale[] PROGMEM = {239,226,213,201,190,179,169,160,151,142,134,127};

void note (int n, int octave) {
  int prescaler = 8 + Clock - (octave + n/12);
  if (prescaler<1 || prescaler>15 || octave==0) prescaler = 0;
  DDRB = (DDRB & ~(1<<Output)) | (prescaler != 0)<<Output;
  OCR1C = pgm_read_byte(&scale[n % 12]) - 1;
  GTCCR = (Output == 4)<<COM1B0;
  TCCR1 = 1<<CTC1 | (Output == 1)<<COM1A0 | prescaler<<CS10;
}
```
My code plays a boot-up sound using this routine and plays a tone whenever a button is being pressed. I'm using this kind of routine because it is non-blocking (there are other things going on in the loop() ) and because the standard `tone()` function can play the boot-up sound but not the tone, which is toggled by a button that has a condition linked to an ISR call:

```
attachInterrupt(0, hornISR, FALLING);

void loop () {
if (hornButtonPressed == true) {  
  note(0, 5);
}

if (digitalRead(hornButton) == HIGH && hornButtonPressed == true) {
  note(0, 0);
  hornButtonPressed = false;
}
}

void hornISR() {
  hornButtonPressed = true;
}

```
As soon as I call `anyrtttl::blocking::playProgMem(BUZZER_PIN, pacman);`, the tune plays, but the tone routine is unable to play anymore. The anyrtttl library is being used in blocking mode because there is no need to run other things while the tune is playing, however, everything should return to normal once the tune has stopped playing.

end2endzone, 2019-08-18:
I am not familiar with arduino registers that you are using in your `note()` function. I don't know what they are actually modifying. I am assuming that you are using the timers. The `playProgMem()` function is using by default the standard arduino `tone()` and `noTone()` functions which uses the Timer2 (see https://forum.arduino.cc/index.php?topic=448977.0). If your function or the `tone()` and `noTone()` are disturbing the default behavior of the timers, it will prevent the melody.

end2endzone, 2019-08-18:
My suggestion would be to try to read and "save" all the registers you are using in your `note()` function. Then play a melody with the `anyrtttl::blocking::playProgMem()` function. Once the melody is done playing read the same registers again and see if any values have changed. This would indicate that `tone()` or `noTone()` function are modifying them.

Keep me informed. If you find any issues with the library (like if you find what it is not "restoring back to normal" once it is done playing, please provide a pull request to correct the situation.

end2endzone, 2019-08-23:
did it help solving your issue?

JeanHenriqueAlves, 2019-08-25:
> My suggestion would be to try to read and "save" all the registers you are using in your `note()` function. Then play a melody with the `anyrtttl::blocking::playProgMem()` function. Once the melody [...]

Well, I took a look at the `.cpp` and `.h` files of the library and I didn't see anything that could be causing a problem, since I am able to call `tone()` and `noTone()` in the `setup()` and not have any problems with the `note()` function afterwards (I'm still not sure about calling it the `loop()`, and by the way, I'm not the author of that function). 

For now, I'm closing this issue since I'm not going to use ATtiny85 in my project anymore (I'm
 already running out of programming space). I'm going to switch to a Pro Mini, and if the problem persists, I'll open a new issue. 

Thanks for your support. Your library is awesome, by the way. I'm looking forward to a `playProgMem16Bits()` or a `playProgMem10Bits()` function. And maybe you could even incorporate the `note()` function in your library instead of the `tone()`, since it saves up to 99% of programming space for short note sequences and up to 30% in long note sequences, also, notes can be stopped easily with `note(0, 0)`.

Take a look here: [www.technoblogy.com/show?19YT](url)
It shows a `note()` function for Arduino Uno/Nano/Pro Mini (ATmega328P) and one for Arduino Leonardo/Pro Micro (ATmega32u4).

closed 2019-08-25 by JeanHenriqueAlves