end2endzone/AnyRtttl #4: error compiling example code

alwaysmohawk, 2023-03-11:
getting a compile error:

Invalid conversion from 'void ()(uint8_t, unsigned int, long unsigned int)' {aka 'void ()(unsigned char, unsigned int, long unsigned int)'} to 'anyrtttl::ToneFuncPtr' {aka 'void (*)(unsigned char, short unsigned int, unsigned int)'} [-fpermissive]
ToneFuncPtr _tone = &tone;

when trying to compile the basic example, pulled latest from arduino ide, v 2.2.0

end2endzone, 2023-03-11:
Hi. Thank you for reporting this issue.
According to the error code you got, the signature of the `tone()` function in Arduino seems to have changed. AnyRtttl library is expecting a signature that matches:
```cpp
void (*)(unsigned char, short unsigned int,      unsigned int)
```
but your tone function signature is:
```cpp
void ()(unsigned char,       unsigned int, long unsigned int)
```
I will review this and I will try to find why/where it has changed.

The issue could also be related to the board you are using. I have only tested the library with *Uno* and *Nano*. It is possible that the tone function signature changes with a special board. That is a possibility. If this is the case, please provide the name of the board you are using. I will find the matching macro definition that matches it and I will add support for it.

I am unfamiliar with Arduino 2.2.0. Is this a nightly build ? The current latest (official) released version of Arduino IDE on [its web site](https://www.arduino.cc/en/software) is 2.0.4, not 2.2.0.

As a temporary workaround, you can use Arduino 1.8.x to build and use AnyRtttl. Unless your project has a requirement to use 2.2.0, you should be fine. If you go to the official Arduino IDE download page and scroll to the [Legacy](https://www.arduino.cc/en/software#legacy-ide-18x) section, you can still download version 1.8.19. The [last tests of AnyRtttl](https://ci.appveyor.com/project/end2endzone/anyrtttl/builds/19674336) were executed on Arduino 1.8.4. Its old, I know but I am pretty sure the signature has not changed in the 1.8.x life span.

end2endzone, 2023-03-12:
On Arduino IDE 1.8.13 on Windows, the definition of tone function is specified in file `C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\Arduino.h` and defined as
```cpp
void tone(uint8_t _pin, unsigned int frequency, unsigned long duration = 0);
```

end2endzone, 2023-03-13:
Hi. I think I have solved the issue. Could you please confirm that it is working on your side before I create an official release ?
To do so, execute the following actions:

1. Delete your previous version of the library to make sure you are using a new version. On Windows, you need to navigate to directory `C:\Users\%USERNAME%\Documents\Arduino\libraries` and delete the directory called `AnyRtttl`.
2. Download the new unofficial library version from https://github.com/end2endzone/AnyRtttl/archive/refs/heads/master.zip
3. Open Arduino IDE
4. Go to menu Sketch, Include Library, Add .ZIP library.
5. Navigate to your _Downloads_ directory and select the new patched library .zip file called `AnyRtttl-master.zip`. 
6.  Try building the examples again.

On a side note, I may have found the reason you originally open the issue on end2endzone/SoftTimers. The "More info" link of the library was pointing to SoftTimers repository and not to this one. More details available if you look at commit b0086da23a143abbb26a91228f6e4f10a336048d. The issue should be resolved now. 

I will wait for your confirmation that it is working on your side before creating an official new version. Again thank you for providing feedback. It is much appreciated.

end2endzone, 2023-03-18:
@alwaysmohawk Did you had time to look at the correction to see if it resolves the issue ?

end2endzone, 2023-03-27:
I have released version [2.2.1](https://github.com/end2endzone/AnyRtttl/releases/tag/2.2.1).