end2endzone/AnyRtttl #9: Question - can we target two speakers ?

lutin-malin, 2025-04-15:
I'm experimenting your library, which is very effective on a single speaker.
I'm not a C++ expert, Is there any way to adress two different speakers in parallel ?

My use case is an nice arcade game made of buttons, leds, and one speaker per player.
I'd have loved to be abled to adress each speaker independantly and play RTTTL songs per player depending on their progress in the versus game they are playing.

Any ideas ?

Thanks

end2endzone, 2025-04-15:
Salut lutin-malin. The answer is no. You could play on 2 speakers but one at a time, not simultaneously. It was not designed in this way. The library stores the current melody "state" in [these global variables](https://github.com/end2endzone/AnyRtttl/blob/2.3.0/src/anyrtttl.cpp#L29-L42). Because of their global nature, starting a new melody would override the state the the currently playing melody.

To support such a feature, I would need to move these variables to a common struct and have the sketch author create 2 different "state" instances. Each function would need to be modified to take the "state structure" as input to know which copy of the state it applies to. For example `isPlaying()` would have to be changed to `isPlaying(&speaker1_state)` to properly report the current state of the speaker 1.

If you still want to try to play on 2 speakers (one at a time), please keep me posted. To be honest, I have never tried. It is not a conventional use case. I would love to know about it if you go this route. The library is mostly used to play melodies. If a single player can hear 2 melodies playing at the same time, it might not be the most pleasant experience. However, it you use these 2 piezo speakers to play sounds (mostly short burst of small RTTTL melodies) I think it would work great.

You could leave this issue opened and if I ever go back to the library, I could implement such a feature. But that most unlikely that I will have time to improve the library.

end2endzone, 2025-04-17:
I apologize for giving you this impression. I agree the sentence can be a bit misleading and there are multiple ways it can be interpreted. It was not written with bad intentions. I will leave this issue open and will eventually try to fix the wording to make it more obvious that it can't be use simultaneously.

> (i guess the internal timer used are the same).

Yes you are right. Even if I would fix the library, you still need a hardware platform (microcontroller) that can use two different timer for each pins connected to speakers.

> To confirm, do you think i can try to edit your current code from namespace to class, and enable to instanciante two objects with different pins (and move the global vars to private).

I'm really sorry but I can't. I do not have as much free time as I used to for my hobbies and have other priorities.

> I'll try to go this way for now, unless you tell me it is vain.

Here are two idea that I think are worth exploring to solve this:
1. Try to use the tone library to play two tones at the same time. This would validate that your microcontroller has indeed multiple timers. A quick google search for `how to play two tone simultaneously on two different pins with esp32 and piezo buzzer` has given the following _**untested**_ code:
```cpp
// Define the pins for the buzzers
const int BUZZER_PIN1 = 26;
const int BUZZER_PIN2 = 27;

// Define the PWM channels
const int TONE_PWM_CHANNEL_1 = 0;
const int TONE_PWM_CHANNEL_2 = 1;

void setup() {
  // Attach the buzzers to their respective PWM channels
  ledcAttachPin(BUZZER_PIN1, TONE_PWM_CHANNEL_1);
  ledcAttachPin(BUZZER_PIN2, TONE_PWM_CHANNEL_2);
}

void loop() {
  // Play a note on buzzer 1
  ledcWriteNote(TONE_PWM_CHANNEL_1, NOTE_C, 4);

  // Play a different note on buzzer 2
  ledcWriteNote(TONE_PWM_CHANNEL_2, NOTE_G, 4);

  delay(500);
}
```
It might be worth a try.

2. You can "duplicate" the library (calling it AnnyRtttl2) easily and have two independent library in your code. Each with their own set of variables. This would allow you to control individually two pins simultaneously.
    1. Find the location where the arduino libraries are installed on your system. On Windows it is located in `C:\Users\%USERNAME%\Documents\Arduino\libraries\AnyRtttl`.
    2. Copy and paste the original AnyRtttl directory giving it a new name (for example AnyRtttl2). 
    3. In AnyRtttl2 directory, edit file the copied library, edit the file `library.properties` and change the name of the library to AnyRtttl2.
    4. In the directory AnyRtttl2/src, edit fhe *.c and *.h files and replace `namespace anyrtttl` with `namespace anyrtttl2`. This will essentially prevent contamination from one library to the other.
    5. Then write your code so that speaker 1 uses native AnyRtttl calls, and have speaker 2 uses calls from AnyRtttl2.
3. Even with option 2, you would still have to assign a different Tone() function to each library to force each library to use a different timer. See [examples/ESP32Rtttl/ESP32Rtttl.ino](https://github.com/end2endzone/AnyRtttl/blob/master/examples/ESP32Rtttl/ESP32Rtttl.ino) for an example.

end2endzone, 2025-10-31:
Hi @lutin-malin 
I am about to update the library with new features.
One of them is that I have implemented support for playing RTTTL melodies on 2 speakers simultaneously.

end2endzone, 2025-10-31:
Please be patient. I will release a new version of the libary soon.
The PR will however close this issue.

You could test the feature early by [downloading the library as a zip](https://github.com/end2endzone/AnyRtttl/archive/refs/heads/master.zip). It will still be identified as version 2.4.0 but the feature is definitely there! There is also an example as well: [ESP32DualPlayRtttl.ino](https://github.com/end2endzone/AnyRtttl/tree/2834273f8884dc774b1c99cf22bd10b2b6838d96/examples/ESP32DualPlayRtttl).

It you like the library, consider leaving a star!